EVerest/EVerest #471: Using Ninja as the Cmake generator

A-thaVan, 2023-12-08:
I had trouble compiling this repo in VSCODE and the Cmake Tools extension due to its preference for Ninja.  

However, I found by running `export CMAKE_GENERATOR=Ninja` before `cmake ..`, I was able to build in VSCODE.

Would the team consider adding this line of code (as optional) to the build documentation for others who wish to use this build environment?

SebaLukas, 2023-12-11:
Hi :)

you can also add `-GNinja` to `cmake ..` -> `cmake .. -GNinja`
With `-G` you can specify your build [system](https://cmake.org/cmake/help/latest/manual/cmake.1.html#cmdoption-cmake-G).

If someone wants to use a different build program other than `make`, that's no problem.
But we can't add the specific commands for every build program in our documentation.

A-thaVan, 2023-12-11:
That is fair.  I have also found that Ninja does not consolidate the binaries, libraries, and various files into the bin, lib, libexec, share, and etc subfolders in build/dist like the scripts in run-scripts are expecting.   CMAKE_RUNTIME_OUTPUT_DIRECTORY CMAKE_LIBRARY_OUTPUT_DIRECTORY only redirects binaries and libraries.  I think we are stuck with Unix Makefiles for the time being.

closed 2023-12-11 by SebaLukas