EVerest/EVerest #510: Plug unlocked before contactor is open after transition from CP state E to C

lategoodbye, 2024-01-25:
Hi,
we have recently implemented the new connector lock interface on our Tarragon platform (everest-core 2023.12.0). During our tests with AC free charging (Dummy token provider), we noticed an unexpected behavior. In case we are in free charging (CP state C, contactor closed) and then switch to CP state E, the plug lock is unlock before the contactor is open. So it seems the EvseManager has an issue.

Expected behavior:
For safety reason the plug lock should only be opened after the contactor open.

Observed behavior:
After a CP state change from E to C, the plug lock is opened before the contactor is open.

Log:
````
Jan 23 14:55:20 tarragon manager[7678]: 2024-01-23 14:55:20.465163 [INFO] connector_1:Evs  ::                                     CAR IEC Event CarRequestedPower
Jan 23 14:55:20 tarragon manager[7678]: 2024-01-23 14:55:20.466460 [INFO] connector_1:Evs  :: EVSE IEC Charger state: PrepareCharging->Charging
Jan 23 14:55:20 tarragon manager[7681]: 2024-01-23 14:55:20.553929 [INFO] tarragon_bsp:Cb  :: Closing contactor...
Jan 23 14:55:20 tarragon manager[7681]: 2024-01-23 14:55:20.646881 [INFO] tarragon_bsp:Cb  :: Contactor state: CLOSED
Jan 23 14:55:20 tarragon manager[7678]: 2024-01-23 14:55:20.680267 [INFO] connector_1:Evs  :: EVSE IEC Event PowerOn
Jan 23 14:55:25 tarragon manager[7681]: 2024-01-23 14:55:25.892872 [INFO] tarragon_bsp:Cb  :: CP state change from C to E, U_CP+: 764 mV, U_CP-: -11175 mV
Jan 23 14:55:26 tarragon manager[7677]: 2024-01-23 14:55:26.555977 [INFO] cb_tarragon_plu  :: is unlocked
Jan 23 14:55:26 tarragon manager[7681]: 2024-01-23 14:55:26.659319 [INFO] tarragon_bsp:Cb  :: Opening contactor...
Jan 23 14:55:26 tarragon manager[7681]: 2024-01-23 14:55:26.691790 [INFO] tarragon_bsp:Cb  :: Contactor state: OPEN
Jan 23 14:55:26 tarragon manager[7678]: 2024-01-23 14:55:26.732036 [INFO] connector_1:Evs  :: EVSE IEC Event PowerOff
````

Config:
````
active_modules:
  api:
    module: API
    connections:
      evse_manager:
      - module_id: connector_1
        implementation_id: evse
  auth:
    module: Auth
    config_module:
      connection_timeout: 10
      prioritize_authorization_over_stopping_transaction: true
      selection_algorithm: PlugEvents
    connections:
      token_provider:
      - module_id: token_provider_1
        implementation_id: main
      token_validator:
      - module_id: token_validator
        implementation_id: main
      evse_manager:
      - module_id: connector_1
        implementation_id: evse
  cb_tarragon_driver:
    module: CbTarragonDriver
    connections: {}
  cb_tarragon_pluglock:
    module: CbTarragonPlugLock
    connections: {}
  energy_manager:
    module: EnergyManager
    connections:
      energy_trunk:
      - module_id: grid_connection_point
        implementation_id: energy_grid
  connector_1:
    module: EvseManager
    config_module:
      ac_enforce_hlc: false
      ac_hlc_enabled: false
      ac_hlc_use_5percent: true
      ac_nominal_voltage: 230
      ac_with_soc: false
      charge_mode: AC
      connector_id: 1
      country_code: DE
      dbg_hlc_auth_after_tstep: false
      ev_receipt_required: false
      evse_id: DE*CHB*E123456*1
      has_ventilation: true
      max_current_import_A: 16
      payment_enable_contract: false
      payment_enable_eim: true
      session_logging: false
      session_logging_path: /tmp/everest-logs
      session_logging_xml: false
      three_phases: true
    connections:
      bsp:
        - module_id: cb_tarragon_driver
          implementation_id: evse_board_support
      connector_lock:
        - module_id: cb_tarragon_pluglock
          implementation_id: connector_lock
  grid_connection_point:
    module: EnergyNode
    config_module:
      fuse_limit_A: 63
      phase_count: 3
    connections:
      energy_consumer:
      - implementation_id: energy_grid
        module_id: connector_1
  token_provider_1:
    module: DummyTokenProvider
    config_implementation:
      main:
        timeout: 10
        token: DEADBEEF
        type: RFID
    connections:
      evse:
      - module_id: connector_1
        implementation_id: evse
  token_validator:
    module: DummyTokenValidator
    config_implementation:
      main:
        sleep: 0.25
        validation_reason: Token seems valid
        validation_result: Accepted
    connections: {}
x-module-layout: {}
````

barsnick, 2024-01-26:
@corneliusclaussen, could PR https://github.com/EVerest/everest-core/pull/511 affect or even fix this?

corneliusclaussen, 2024-01-26:
> @corneliusclaussen, could PR #511 affect or even fix this?

No, the (mispelled) name of PR511 is about mutex locking and not about motor lock. I'll have a look into why the order is wrong here.

corneliusclaussen, 2024-02-01:
This PR contains a fix. It has not been tested yet, but it compiles ;-)
Could you test it in your setup and see if it fixes your issue?

https://github.com/EVerest/everest-core/pull/522

lategoodbye, 2024-02-01:
I tested an build including #522 and the issue is resolved:
````
Feb 01 14:25:25 tarragon manager[530]: 2024-02-01 14:25:25.118539 [INFO] tarragon_bsp:Cb  :: PP ampacity change from None to A_13 (U_PP: 2089 mV)
Feb 01 14:25:35 tarragon manager[530]: 2024-02-01 14:25:35.325602 [INFO] tarragon_bsp:Cb  :: CP state change from A to B, U_CP+: 8928 mV, U_CP-: -2 mV
Feb 01 14:25:35 tarragon manager[531]: 2024-02-01 14:25:35.974096 [INFO] tarragon_pluglo  :: plug is locked. Feedback voltage: 626 mV
Feb 01 14:25:36 tarragon manager[530]: 2024-02-01 14:25:36.205669 [INFO] tarragon_bsp:Cb  :: Read PP ampacity value: A_13 (U_PP: 2089 mV)
Feb 01 14:25:36 tarragon manager[530]: 2024-02-01 14:25:36.345263 [INFO] tarragon_bsp:Cb  :: handle_pwm_on: Setting new duty cycle of 21.67%
Feb 01 14:25:40 tarragon manager[530]: 2024-02-01 14:25:40.818785 [INFO] tarragon_bsp:Cb  :: CP state change from B to C, U_CP+: 5920 mV, U_CP-: -11865 mV
Feb 01 14:25:41 tarragon manager[530]: 2024-02-01 14:25:41.044138 [INFO] tarragon_bsp:Cb  :: Closing contactor...
Feb 01 14:25:41 tarragon manager[530]: 2024-02-01 14:25:41.044626 [INFO] tarragon_bsp:Cb  :: Contactor state: CLOSED
Feb 01 14:25:50 tarragon manager[530]: 2024-02-01 14:25:50.004330 [INFO] tarragon_bsp:Cb  :: CP state change from C to E, U_CP+: 755 mV, U_CP-: -11855 mV
Feb 01 14:25:50 tarragon manager[530]: 2024-02-01 14:25:50.252793 [INFO] tarragon_bsp:Cb  :: Opening contactor...
Feb 01 14:25:50 tarragon manager[530]: 2024-02-01 14:25:50.253239 [INFO] tarragon_bsp:Cb  :: Contactor state: OPEN
Feb 01 14:25:50 tarragon manager[531]: 2024-02-01 14:25:50.904305 [INFO] tarragon_pluglo  :: plug is unlocked. Feedback voltage: 3000 mV
````

Thanks

corneliusclaussen, 2024-02-01:
I will mark the PR as ready for review then

corneliusclaussen, 2024-02-01:
merged into main

closed 2024-02-01 by corneliusclaussen