EVerest/EVerest #529: Segmentation fault (sigsegv) -- set_root_schema -- JSON validator

mmoses0, 2024-02-05:
The source code is cross compiled for the raspberry pi with the bullseye toolchain (armv8-rpi4-linux-gnueabihf). The build is success, however, while running the everest manager, a segmentation fault is observed. I traced the segmentation fault to the function call: validator.set_root_schema(schema)
The input argument is a json file. Attached is the file being read by the software
[validator.txt](https://github.com/EVerest/everest-core/files/14172155/validator.txt)

The runtime failure is: 
 everest-dev.service: Main process exited, code=dumped, status=11/SEGV
 systemd[1]: everest-dev.service: Failed with result 'core-dump'.

corneliusclaussen, 2024-02-09:
Which version of everest-core are you using?

mmoses0, 2024-02-09:
> Which version of everest-core are you using?

I am using everest-core 2024.1.0. I also attached my workspace config file.

[workspace-config - Copy.txt](https://github.com/EVerest/everest-core/files/14222236/workspace-config.-.Copy.txt)

mmoses0, 2024-02-19:
@corneliusclaussen Do you have any insights?

corneliusclaussen, 2024-02-19:
Which Debian are you running on your pi? Bullseye?

mmoses0, 2024-02-19:
> Which Debian are you running on your pi? Bullseye?

@corneliusclaussen Yes I am running Bullseye.

closed 2025-08-09 by Pietfried