EVerest/EVerest #544: EvseV2G: Integration of libcbv2g as OpenV2G replacement

SebaLukas, 2024-02-13:
Currently the EvseV2G module uses the [OpenV2G](https://github.com/Everest/ext-openv2g) project to encode and decode EXI messages.
The goal now is to replace OpenV2G with the new [libcbv2g](https://github.com/Everest/libcbv2g) library.

The libcbv2g contains the generated c EXI code from the [cbexigen](https://github.com/Everest/cbexigen) generator with a CMake integration.

Tasks:
- [ ] [dependencies.yaml](https://github.com/EVerest/everest-core/blob/main/dependencies.yaml) + [module-dependencies.cmake](https://github.com/EVerest/everest-core/blob/main/module-dependencies.cmake): replace openv2g with libcbv2g
- [ ] EvseV2G module: Customize CMakeLists.txt
- [ ] EvseV2G module: Customize include headers
- [ ] EvseV2G module: Customize src code
- [ ] DIN70121 is working
- [ ] ISO15118-2 AC EIM is working 
- [ ] ISO15118-2 DC EIM is working
- [ ] ISO15118-2 PnC is working

Every contribution is appreciated :smiley:

SiebrenW, 2024-02-21:
I am somewhat prepared to work on this a bit in my free time.
I have no facilities to test this though, besides unit tests and such.

SebaLukas, 2024-02-22:
That's great :+1:  For now, it would be enough if everest-core (EvseV2G module with libcbv2g) builds without errors and warnings. DIN70121 would also be the "easiest" to start with, as PnC is not possible here.
You can use the various SILs for testing.

If you need any help, let me know.

SiebrenW, 2024-02-23:
As mentioned in the cbexigen issue, the SignedInfo element must be encoded in another schema, so the fragments in ISO-2 and ISO-20 for SignedInfo are useless.
This is a bit of a blocker for removing OpenV2G altogether (I started with removing it completely, only to notice that this schema was missing). More in issue EVerest/cbexigen#69

SiebrenW, 2024-02-27:
So the previous isn't an issue after all. Apparently cbv2g decided to duplicate the xmldsig fragment encoders for every message namespace, which on one hand is duplicate logic, but on the other simplifies things (just assign the element instead of converting it from one type to the other.

This PR has the initial changes to remove openv2g: #563

SebaLukas, 2024-06-24:
Done and merged #563

closed 2024-06-24 by SebaLukas