EVerest/EVerest #579: EVSE Paused when using OCPP

ziyadtarek7, 2024-03-06:
Hello,

I have to similar configs where one of the uses dummytokenvalidator while the other uses OCPP,
When ever I try to use the ocpp one I get EVSE paused almost immediately after clicking car plugin in the simulator, while when using the dummytokenvalidator, it works perfectly.
when I first created the ocpp config file it was working as expected but out of nowhere it stopped.

Here is the log in both cases:

dummytokenvalidator:

>  2024-03-06 14:19:51.916337 [INFO] DummyTokenValid  :: Got validation request for token: 0123456789ABCD
2024-03-06 14:19:52.416672 [INFO] DummyTokenValid  :: Returning validation status: Accepted
2024-03-06 14:19:52.439238 [INFO] auth:Auth        :: Providing authorization to connector#1
2024-03-06 14:19:52.769891 [INFO] evse_manager:Ev  :: EVSE IEC Set PWM On (5.000000074505806%) took 0 ms
2024-03-06 14:19:52.915920 [INFO] auth:Auth        :: Result for token: 0123456789ABCD: ACCEPTED
2024-03-06 14:19:52.977889 [INFO] evse_manager:Ev  :: EVSE IEC EIM Authorization received
2024-03-06 14:19:52.978091 [INFO] evse_manager:Ev  :: EVSE IEC Transaction Started (0 kWh)
2024-03-06 14:19:52.978408 [INFO] evse_manager:Ev  :: EVSE IEC DC mode. We are in 5percent mode so we can continue without further action.
2024-03-06 14:19:52.978506 [INFO] evse_manager:Ev  :: EVSE IEC Charger state: Wait for Auth->PrepareCharging
2024-03-06 14:19:54.115482 [INFO] evse_manager:Ev  :: EVSE ISO SLAC MATCHED
2024-03-06 14:19:54.135855 [INFO] evse_manager:Ev  :: EVSE ISO D-LINK_READY (true)

OCPP: 

> 2024-03-06 14:17:44.156164 [INFO] auth:Auth        :: Providing authorization to connector#1
2024-03-06 14:17:44.940664 [INFO] evse_manager:Ev  :: EVSE IEC Set PWM On (5.000000074505806%) took 0 ms
2024-03-06 14:17:45.130324 [INFO] auth:Auth        :: Result for token: 0123456789ABCD: ACCEPTED
2024-03-06 14:17:45.206881 [INFO] evse_manager:Ev  :: EVSE IEC EIM Authorization received
2024-03-06 14:17:45.207509 [INFO] evse_manager:Ev  :: EVSE IEC Transaction Started (0 kWh)
2024-03-06 14:17:45.208729 [INFO] evse_manager:Ev  :: EVSE IEC DC mode. We are in 5percent mode so we can continue without further action.
2024-03-06 14:17:45.209089 [INFO] evse_manager:Ev  :: EVSE IEC Charger state: Wait for Auth->PrepareCharging
2024-03-06 14:17:45.287327 [INFO] OCPP0:OCPP       :: EVSE#1: Received TransactionStarted
2024-03-06 14:17:45.616218 [INFO] evse_manager:Ev  :: EVSE IEC Transaction Finished: DeAuthorized (0 kWh)
2024-03-06 14:17:45.710445 [INFO] evse_manager:Ev  :: EVSE IEC Charger state: PrepareCharging->EVSE Paused
2024-03-06 14:17:45.710753 [INFO] evse_manager:Ev  :: EVSE IEC Set PWM Off
2024-03-06 14:17:46.126347 [INFO] evse_manager:Ev  :: EVSE ISO SLAC MATCHED
2024-03-06 14:17:46.146723 [INFO] evse_manager:Ev  :: EVSE ISO D-LINK_READY (true)

Any help would be appreciated, Thank you!

corneliusclaussen, 2024-09-05:
Is this still an issue or can we close it?

Pietfried, 2024-09-05:
The reason for the issue is that the transaction was not accepted by the CSMS if you use the dummy token validator, probably becasue the token is not known to the CSMS. This causes the EVSE to pause.

closed 2024-09-05 by Pietfried