EVerest/EVerest #812: Target "framework" links to target "ryml::ryml" but the target was not   found.

shengxiang200788, 2024-08-02:
### Describe the bug

CMake Error at /home/eepp/checkout/everest-workspace/everest-framework/lib/CMakeLists.txt:10 (add_library):
  Target "framework" links to target "ryml::ryml" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

### EVerest Domain

Other

Pietfried, 2024-08-05:
Hi @shengxiang200788 , thanks for reporting this issue. Please add information in the **To Reproduce** and include things like EVerest version, compile options and your system information.

Best
Piet

Pietfried, 2025-08-11:
Closing due to inactivity. Please reopen if you have more details to share.

closed 2025-08-11 by Pietfried