EVerest/EVerest #829: EvseManager: `allow_power_on(false)` gets called repeatedly while error active

a-w50, 2024-08-13:
### Describe the bug

Given, that
-  the `EvseManager` is in state `PrepareCharging`
- and an error gets raised, which prevents charging
then, the `EvseManager` continuously calls `allow_power_on(false)`, due to this function:
```
bool Charger::errors_prevent_charging_internal() {
    if (shared_context.error_prevent_charging_flag) {
        graceful_stop_charging();
        return true;
    }
    return false;
}
```
(Note, the name and signature of this function suggests, that it only checks something - but it has *side-effects*, i.e. `graceful_stop_charging()`)

I would expect that `allow_power_on(false)` gets called only as long as the bsp event `PowerOff` hasn't been received and probably at a lower rate (`MAINLOOP_UPDATE_RATE` = 100ms right now).

### EVerest Domain

Charge Control

### Affected EVerest Module

EvseManager

### To Reproduce

Get the `EvseManager` into state `PrepareCharging` and raise a `VendorError` from the `board_support` module.

dev-enphase, 2024-08-20:
Is there any traction on this issue? Please fix this issue ASAP as this is causing performance issue in the system

closed 2024-09-05 by corneliusclaussen

dev-enphase, 2024-09-05:
> ZjQcmQRYFpfptBannerStart
> This Message Is From an External Sender
> Do not click on any links, reply or open attachments unless you recognize the sender and know the content is safe. If suspicious report to ***@***.***
>
> ZjQcmQRYFpfptBannerEnd
>
> Closed #829<https://urldefense.com/v3/__https://github.com/EVerest/everest-core/issues/829__;!!J0D89H32QIeNiQ!zwfr5Z_5Fh9v7taNTgW5ovMzlrwaGMpwnIRXTjYAbeGfb89zw9qpEXYkxz0rafGbfM8Z4ZUWPhiiHGFKB5pUMXk1$> as completed via #849<https://urldefense.com/v3/__https://github.com/EVerest/everest-core/pull/849__;!!J0D89H32QIeNiQ!zwfr5Z_5Fh9v7taNTgW5ovMzlrwaGMpwnIRXTjYAbeGfb89zw9qpEXYkxz0rafGbfM8Z4ZUWPhiiHGFKB9ih5tF0$>.

[like]  Devendra Chouhan reacted to your message: