EVerest/EVerest #922: Node version error for Node-red UI table for nodered-sil.sh script

luxscious, 2024-10-18:
### Describe the bug

I tried to follow the tutorial for getting setup on everest quickly (I followed this a couple of months ago with success), but I'm encountering errors with the nodered-sil.sh script. I believe the new version of node-red needs Node version >=16 however the docker image that this script is pulling from uses version ~14. 

[image]

Basically, the ui-table element is non-usable which breaks the node-red ui entirely. 

### EVerest Domain

Simulation

### Affected EVerest Module

docker image: everest-nodered

script: /everest/everest-core/build/run-scripts/nodered-sil.sh

### To Reproduce

follow the instructions to get setup on everest: (https://everest.github.io/latest/general/03_quick_start_guide.html)
This happens when running:
{EVerest Workspace Directory}/everest-core/build/run-scripts/nodered-sil.sh

### Anything else?

I believe you just need to update the Node version in the docker image as the ui table works fine for the other nodered image.

mayuresh-savargaonkar, 2024-10-27:
Was there a fix to this?

closed 2024-10-30 by Pietfried

paulwuertz, 2024-11-11:
I also saw the same issue. Following the guide leaves me with an empty ui sub-page.

[image]

While executing `./user_workspace/everest-core/build/run-scripts/nodered-sil.sh` I also get the warning about

> notsup Unsupported engine for node-red-node-ui-table@0.4.4: wanted: {"node":">=16"} (current: {"node":"14.19.3","npm":"6.14.17"})

[image]

 ENV NODE_VERSION=14.19.3 is set in the [node-red container](nodered/node-red:2.2.3-minimal)

paulwuertz, 2024-11-11:
Related to https://github.com/EVerest/everest-core/pull/895.
Adding RUN npm install node-red-node-ui-table**@0.4.3** on the build Dockerfile at everest-core/build/dist/share/everest/docker/Dockerfile made the dashboard work ^^