EVerest/EVerest #978: Withdraw authorization

Pietfried, 2024-12-03:
### Describe the problem

The [Auth module](https://github.com/EVerest/everest-core/tree/main/modules/Auth) is a central module in EVerest that is able to handle incoming authorization requests from modules implementing the `token_provider` interface and forwards these requests to modules implemeting the `token_validator` interface.

If an authorization request was accepted, the Auth module provides authorization via the `authorize_response` command of the `evse_manager` requirement(s).

Currently there is no way to withdraw this granted authorization. The authorization can only timeout. A valid use case for authorization terminals and displays is to be able to withdraw authorization which was previously requested or granted.

### EVerest Domain

Authorization

### Affected EVerest Module

Auth

### Describe your solution

The goal of this issue is to extend the `auth` interface by an additional command `withdraw_authorization` and to implement this command as part of the `Auth` module. The implementation of the command shall withdraw the specified granted authorization based on the specified parameters.

The parameters of the command shall be:
* evse_id: int 
* id_token: [IdToken](https://everest.github.io/nightly/_generated/types/authorization.html#authorization-idtoken)

If only the evse_id is given, the granted authorization for this EVSE shall be withdrawn.
If only the id_token is given, the granted authorization for every EVSE where this id_token is placed shall be withdrawn
If both parameters are given, the granted authorization for the given EVSE shall be withdrawn, if the placed id_token matches the given id_token
If no parameter is given, all granted authorizations for all EVSEs shall be removed  

Note that it should be possible to withdraw authorization in all of the following phases:
* Authorization was requested but no EVSE has yet been selected
* Authorization was requested, EVSE is selected but EV has not yet been plugged in
* Authorization was requested, EVSE is selected and EV is plugged in (transaction is running) 

In the last two cases the `withdraw_authorization` command at the `evse_manager` requirement shall be called as part of handling the new command. 
 

### Additional context

This likely requires these (or similar) types definitions:

```
WithdrawAuthorizationRequest {
evse_id: int (optional)
id_token: IdToken (optional)
}

WithdrawAuthorizationResult {
ACCEPTED, // in case requested authorization was removed 
AUTHORIZATION_NOT_FOUND, // in case no match for request was found
REJECTED // in case module could not process the request for other reasons
}

corneliusclaussen, 2024-12-04:
"Authorization was requested, EVSE is selected and EV is plugged in (transaction is running)": This case may also require changes in EvseManager. Currently withdraw_authorization() in evse_manager implementation only succeeds if charging was not started yet. This would need to be changed, so that EvseManager ends the transaction when auth is withdrawn.

Pietfried, 2024-12-13:
> "Authorization was requested, EVSE is selected and EV is plugged in (transaction is running)": This case may also require changes in EvseManager. Currently withdraw_authorization() in evse_manager [...]

Since the Auth module has the requirement `evse_manager` it can also directly call the command `stop_transaction`. This is already done by the auth module e.g. in case a token is swiped for a second time to stop a tx

corneliusclaussen, 2024-12-13:
In this case no changes to EvseManager would be neccessary

closed 2025-03-23 by Pietfried