EVerest/EVerest #998: Everest compilation failing due to libevse_security on commit ID d8db94896171a96ef7bcc734338b2c6e59a915b9

dev-enphase, 2024-12-18:
### Describe the bug

We are seeing EVerest core compilation issues on commit id d8db94896171a96ef7bcc734338b2c6e59a915b9. This is failing in 

/home/dev/.cache/CPM/libevse-security/2cbba2d59b84cc27252129605648c046abd91507/libevse-security/3rd_party/cert_rehash/c_rehash.hpp:125:95: error: narrowing conversion of -1 from int to char [-Wnarrowing]

This compilation was working well in October release. 

### EVerest Domain

Compilation

### Affected EVerest Module

Clone repositories using end-init. compile Everest on main. 

### Anything else?

[Uploading CompilationLogs-failure.txt…]()

dev-enphase, 2024-12-18:
@jannejy This compilation issue we are seeing in [d8db94896171a96ef7bcc734338b2c6e59a915b9] where it failing in function of libevese_security 

static int handle_symlink(const char* filename, const char* fullpath) {
    static char xdigit[] = {0,  1,  2,  3,  4,  5,  6,  7,  8,  9,  -1, -1, -1, -1, -1, -1, -1, 10, 11,
                            12, 13, 14, 15, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1,
                            -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, 10, 11, 12, 13, 14, 15};
    char linktarget[NAME_MAX], *endptr;
    unsigned int hash = 0;
    unsigned char ch;
    int i, type, id;
    ssize_t n;
    for (i = 0; i < 8; i++) {
        ch = filename[i] - ‘0’;
        if (ch >= countof(xdigit) || xdigit[ch] < 0)
            return -1;
        hash <<= 4;
        hash += xdigit[ch]; (edited)

dev-enphase, 2024-12-18:
[CompilationLogs-failure.txt](https://github.com/user-attachments/files/18180736/CompilationLogs-failure.txt)

jannejy, 2024-12-18:
If you are able to change in that file:
````
static signed char xdigit[] = {0, 1, 2, 3, 4, 5, 6, 7, 8, 9, -1, -1, -1, -1, -1, -1, -1, 10, 11,
12, 13, 14, 15, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1,
-1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, 10, 11, 12, 13, 14, 15};
````
does this succeed?

dev-enphase, 2024-12-18:
This change is working fine in my local build where I have libocpp and libevse_security available locally. Can you push the change in libevse_security repo and update libocpp/everest-core with the commit.

In our project we are using just everest-core commit id in dependencies.yaml so I need everest-core commit

dev-enphase, 2024-12-23:
This is fix in libevse-security repository

closed 2024-12-23 by dev-enphase