EVerest/EVerest #1014: EvseSlac: No repetition of CM_ATTEN_CHAR.IND in timeout case

lategoodbye, 2025-01-20:
### Describe the bug

According to ISO 15118-3 requirement V2G3-A09-46 the EVSE should repeat up to C_EV_match_retry times the CM_ATTEN_CHAR.IND in case it doesn't receive a CM_ATTEN_CHAR.RSP within the defined timeout.

Everest log (based on EVerest 2024.9.0):
````
2022-04-28T20:04:44.268371+0200 chargesom manager[264]: [INFO] evse_slac:EvseS  :: Session (run_id=AD45CD1DE2988D11, ev_mac=00:01:87:0F:75:D3): initialized, waiting for CM_START_ATTEN_CHAR_IND
2022-04-28T20:04:44.508219+0200 chargesom manager[264]: [INFO] evse_slac:EvseS  :: Session (run_id=AD45CD1DE2988D11, ev_mac=00:01:87:0F:75:D3): received CM_START_ATTEN_CHAR_IND, going to substate SOUNDING
2022-04-28T20:04:44.528211+0200 chargesom manager[264]: [INFO] evse_slac:EvseS  :: Session (run_id=AD45CD1DE2988D11, ev_mac=00:01:87:0F:75:D3): needs to be in state WAIT_FOR_START_ATTEN_CHAR for CM_START_ATTEN_CHAR_IND
2022-04-28T20:04:44.548193+0200 chargesom manager[264]: [INFO] evse_slac:EvseS  :: Session (run_id=AD45CD1DE2988D11, ev_mac=00:01:87:0F:75:D3): needs to be in state WAIT_FOR_START_ATTEN_CHAR for CM_START_ATTEN_CHAR_IND
2022-04-28T20:04:44.568132+0200 chargesom manager[264]: [INFO] evse_slac:EvseS  :: Session (run_id=AD45CD1DE2988D11, ev_mac=00:01:87:0F:75:D3): received CM_MNBC_SOUND_IND
2022-04-28T20:04:44.576280+0200 chargesom manager[264]: [INFO] evse_slac:EvseS  :: Session (run_id=AD45CD1DE2988D11, ev_mac=00:01:87:0F:75:D3): received CM_ATTEN_PROFILE_IND
2022-04-28T20:04:44.588110+0200 chargesom manager[264]: [INFO] evse_slac:EvseS  :: Session (run_id=AD45CD1DE2988D11, ev_mac=00:01:87:0F:75:D3): received CM_MNBC_SOUND_IND
2022-04-28T20:04:44.596262+0200 chargesom manager[264]: [INFO] evse_slac:EvseS  :: Session (run_id=AD45CD1DE2988D11, ev_mac=00:01:87:0F:75:D3): received CM_ATTEN_PROFILE_IND
2022-04-28T20:04:44.628586+0200 chargesom manager[264]: [INFO] evse_slac:EvseS  :: Session (run_id=AD45CD1DE2988D11, ev_mac=00:01:87:0F:75:D3): received CM_MNBC_SOUND_IND
2022-04-28T20:04:44.635981+0200 chargesom manager[264]: [INFO] evse_slac:EvseS  :: Session (run_id=AD45CD1DE2988D11, ev_mac=00:01:87:0F:75:D3): received CM_ATTEN_PROFILE_IND
2022-04-28T20:04:44.648051+0200 chargesom manager[264]: [INFO] evse_slac:EvseS  :: Session (run_id=AD45CD1DE2988D11, ev_mac=00:01:87:0F:75:D3): received CM_MNBC_SOUND_IND
2022-04-28T20:04:44.656285+0200 chargesom manager[264]: [INFO] evse_slac:EvseS  :: Session (run_id=AD45CD1DE2988D11, ev_mac=00:01:87:0F:75:D3): received CM_ATTEN_PROFILE_IND
2022-04-28T20:04:44.667974+0200 chargesom manager[264]: [INFO] evse_slac:EvseS  :: Session (run_id=AD45CD1DE2988D11, ev_mac=00:01:87:0F:75:D3): received CM_MNBC_SOUND_IND
2022-04-28T20:04:44.676274+0200 chargesom manager[264]: [INFO] evse_slac:EvseS  :: Session (run_id=AD45CD1DE2988D11, ev_mac=00:01:87:0F:75:D3): received CM_ATTEN_PROFILE_IND
2022-04-28T20:04:44.688160+0200 chargesom manager[264]: [INFO] evse_slac:EvseS  :: Session (run_id=AD45CD1DE2988D11, ev_mac=00:01:87:0F:75:D3): received CM_MNBC_SOUND_IND
2022-04-28T20:04:44.696294+0200 chargesom manager[264]: [INFO] evse_slac:EvseS  :: Session (run_id=AD45CD1DE2988D11, ev_mac=00:01:87:0F:75:D3): received CM_ATTEN_PROFILE_IND
2022-04-28T20:04:44.708078+0200 chargesom manager[264]: [INFO] evse_slac:EvseS  :: Session (run_id=AD45CD1DE2988D11, ev_mac=00:01:87:0F:75:D3): received CM_MNBC_SOUND_IND
2022-04-28T20:04:44.716238+0200 chargesom manager[264]: [INFO] evse_slac:EvseS  :: Session (run_id=AD45CD1DE2988D11, ev_mac=00:01:87:0F:75:D3): received CM_ATTEN_PROFILE_IND
2022-04-28T20:04:45.107526+0200 chargesom manager[264]: [INFO] evse_slac:EvseS  :: Session (run_id=AD45CD1DE2988D11, ev_mac=00:01:87:0F:75:D3): Sounding not yet complete but timed out, going to sub-state FINALIZE_SOUNDING
2022-04-28T20:04:45.152772+0200 chargesom manager[264]: [INFO] evse_slac:EvseS  :: Session (run_id=AD45CD1DE2988D11, ev_mac=00:01:87:0F:75:D3): Finalize sounding, sending CM_ATTEN_CHAR_IND
2022-04-28T20:04:45.353075+0200 chargesom manager[264]: [INFO] evse_slac:EvseS  :: Session (run_id=AD45CD1DE2988D11, ev_mac=00:01:87:0F:75:D3): Waiting for CM_ATTEN_CHAR_RSP timed out -> failed
2022-04-28T20:04:45.354169+0200 chargesom manager[264]: [INFO] evse_slac:EvseS  :: Entered Failed state
````

Hardware
EVSE: Charge SOM DC (uses Vertexcom MS120x as PLC)
EV: CMS (uses QCA7005 as PLC)

### EVerest Domain

SLAC

### Affected EVerest Module

EvseSlac

### To Reproduce

1. Setup a EV simulator, which doesn't reply to the first CM_ATTEN_CHAR.IND
2. start SLAC process

SebaLukas, 2025-02-27:
Hi :)

please have look at this branch: https://github.com/EVerest/everest-core/tree/fix%2Fevse_slac_no_repetition_cm_atten_char_ind

At the moment I have no possibility to test this changes. Since you have access to a HW EV simulator, could you test these changes for me?

lategoodbye, 2025-02-28:
@SebaLukas I just started a longterm test between a CCC (running EVerest as EVSE) and a CCP (running our EV stack). I hope this reproduce the repetition.

lategoodbye, 2025-02-28:
FWIW, here is a trace were all repetitions failed:

````
2025-02-28T13:53:33.264779+0100 tarragon manager[20413]: [INFO] EvseSlac0:EvseS  :: Session (run_id=5331739AD54C997D, ev_mac=00:01:87:05:A0:84): initialized, waiting for CM_START_ATTEN_CHAR_IND
2025-02-28T13:53:33.289069+0100 tarragon manager[20413]: [INFO] EvseSlac0:EvseS  :: Session (run_id=5331739AD54C997D, ev_mac=00:01:87:05:A0:84): received CM_START_ATTEN_CHAR_IND, going to substate SOUNDING
2025-02-28T13:53:33.326891+0100 tarragon manager[20413]: [INFO] EvseSlac0:EvseS  :: Session (run_id=5331739AD54C997D, ev_mac=00:01:87:05:A0:84): needs to be in state WAIT_FOR_START_ATTEN_CHAR for CM_START_ATTEN_CHAR_IND
2025-02-28T13:53:33.361247+0100 tarragon manager[20413]: [INFO] EvseSlac0:EvseS  :: Session (run_id=5331739AD54C997D, ev_mac=00:01:87:05:A0:84): needs to be in state WAIT_FOR_START_ATTEN_CHAR for CM_START_ATTEN_CHAR_IND
2025-02-28T13:53:33.371394+0100 tarragon manager[20404]: [INFO] token_validator  :: Returning validation status: Accepted
2025-02-28T13:53:33.377274+0100 tarragon manager[20416]: [INFO] auth:Auth        :: Providing authorization to evse#1
2025-02-28T13:53:33.395301+0100 tarragon manager[20413]: [INFO] EvseSlac0:EvseS  :: Session (run_id=5331739AD54C997D, ev_mac=00:01:87:05:A0:84): received CM_MNBC_SOUND_IND
2025-02-28T13:53:33.395301+0100 tarragon manager[20413]: [INFO] EvseSlac0:EvseS  :: Session (run_id=5331739AD54C997D, ev_mac=00:01:87:05:A0:84): received CM_ATTEN_PROFILE_IND
2025-02-28T13:53:33.436232+0100 tarragon manager[20413]: [INFO] EvseSlac0:EvseS  :: Session (run_id=5331739AD54C997D, ev_mac=00:01:87:05:A0:84): received CM_MNBC_SOUND_IND
2025-02-28T13:53:33.436232+0100 tarragon manager[20413]: [INFO] EvseSlac0:EvseS  :: Session (run_id=5331739AD54C997D, ev_mac=00:01:87:05:A0:84): received CM_ATTEN_PROFILE_IND
2025-02-28T13:53:33.465851+0100 tarragon manager[20413]: [INFO] EvseSlac0:EvseS  :: Session (run_id=5331739AD54C997D, ev_mac=00:01:87:05:A0:84): received CM_MNBC_SOUND_IND
2025-02-28T13:53:33.467902+0100 tarragon manager[20413]: [INFO] EvseSlac0:EvseS  :: Session (run_id=5331739AD54C997D, ev_mac=00:01:87:05:A0:84): received CM_ATTEN_PROFILE_IND
2025-02-28T13:53:33.513488+0100 tarragon manager[20413]: [INFO] EvseSlac0:EvseS  :: Session (run_id=5331739AD54C997D, ev_mac=00:01:87:05:A0:84): received CM_MNBC_SOUND_IND
2025-02-28T13:53:33.515257+0100 tarragon manager[20413]: [INFO] EvseSlac0:EvseS  :: Session (run_id=5331739AD54C997D, ev_mac=00:01:87:05:A0:84): received CM_ATTEN_PROFILE_IND
2025-02-28T13:53:33.536900+0100 tarragon manager[20413]: [INFO] EvseSlac0:EvseS  :: Session (run_id=5331739AD54C997D, ev_mac=00:01:87:05:A0:84): received CM_MNBC_SOUND_IND
2025-02-28T13:53:33.538669+0100 tarragon manager[20413]: [INFO] EvseSlac0:EvseS  :: Session (run_id=5331739AD54C997D, ev_mac=00:01:87:05:A0:84): received CM_ATTEN_PROFILE_IND
2025-02-28T13:53:33.572381+0100 tarragon manager[20413]: [INFO] EvseSlac0:EvseS  :: Session (run_id=5331739AD54C997D, ev_mac=00:01:87:05:A0:84): received CM_MNBC_SOUND_IND
2025-02-28T13:53:33.574156+0100 tarragon manager[20413]: [INFO] EvseSlac0:EvseS  :: Session (run_id=5331739AD54C997D, ev_mac=00:01:87:05:A0:84): received CM_ATTEN_PROFILE_IND
2025-02-28T13:53:33.608371+0100 tarragon manager[20413]: [INFO] EvseSlac0:EvseS  :: Session (run_id=5331739AD54C997D, ev_mac=00:01:87:05:A0:84): received CM_MNBC_SOUND_IND
2025-02-28T13:53:33.610222+0100 tarragon manager[20413]: [INFO] EvseSlac0:EvseS  :: Session (run_id=5331739AD54C997D, ev_mac=00:01:87:05:A0:84): received CM_ATTEN_PROFILE_IND
2025-02-28T13:53:33.649842+0100 tarragon manager[20413]: [INFO] EvseSlac0:EvseS  :: Session (run_id=5331739AD54C997D, ev_mac=00:01:87:05:A0:84): received CM_MNBC_SOUND_IND
2025-02-28T13:53:33.649842+0100 tarragon manager[20413]: [INFO] EvseSlac0:EvseS  :: Session (run_id=5331739AD54C997D, ev_mac=00:01:87:05:A0:84): received CM_ATTEN_PROFILE_IND
2025-02-28T13:53:33.678865+0100 tarragon manager[20413]: [INFO] EvseSlac0:EvseS  :: Session (run_id=5331739AD54C997D, ev_mac=00:01:87:05:A0:84): received CM_MNBC_SOUND_IND
2025-02-28T13:53:33.678865+0100 tarragon manager[20413]: [INFO] EvseSlac0:EvseS  :: Session (run_id=5331739AD54C997D, ev_mac=00:01:87:05:A0:84): received CM_ATTEN_PROFILE_IND
2025-02-28T13:53:33.714957+0100 tarragon manager[20413]: [INFO] EvseSlac0:EvseS  :: Session (run_id=5331739AD54C997D, ev_mac=00:01:87:05:A0:84): received CM_MNBC_SOUND_IND
2025-02-28T13:53:33.714957+0100 tarragon manager[20413]: [INFO] EvseSlac0:EvseS  :: Session (run_id=5331739AD54C997D, ev_mac=00:01:87:05:A0:84): received CM_ATTEN_PROFILE_IND
2025-02-28T13:53:33.714957+0100 tarragon manager[20413]: [INFO] EvseSlac0:EvseS  :: Session (run_id=5331739AD54C997D, ev_mac=00:01:87:05:A0:84): received all sounds, going to substate FINALIZE_SOUNDING
2025-02-28T13:53:33.760817+0100 tarragon manager[20413]: [INFO] EvseSlac0:EvseS  :: Session (run_id=5331739AD54C997D, ev_mac=00:01:87:05:A0:84): Finalize sounding, sending CM_ATTEN_CHAR_IND
2025-02-28T13:53:33.961798+0100 tarragon manager[20413]: [INFO] EvseSlac0:EvseS  :: Session (run_id=5331739AD54C997D, ev_mac=00:01:87:05:A0:84): Waiting for CM_ATTEN_CHAR_RSP timed out -> retry matching
2025-02-28T13:53:33.961798+0100 tarragon manager[20413]: [INFO] EvseSlac0:EvseS  :: Session (run_id=5331739AD54C997D, ev_mac=00:01:87:05:A0:84): Finalize sounding, sending CM_ATTEN_CHAR_IND
[... cut ...]
2025-02-28T13:53:34.364535+0100 tarragon manager[20413]: [INFO] EvseSlac0:EvseS  :: Session (run_id=5331739AD54C997D, ev_mac=00:01:87:05:A0:84): Waiting for CM_ATTEN_CHAR_RSP timed out -> failed
2025-02-28T13:53:34.392449+0100 tarragon manager[20413]: [INFO] EvseSlac0:EvseS  :: Session (run_id=2BECC771454FC714, ev_mac=00:01:87:05:A0:84): needs to be in state WAIT_FOR_START_ATTEN_CHAR for CM_START_ATTEN_CHAR_IND
2025-02-28T13:53:34.428855+0100 tarragon manager[20413]: [INFO] EvseSlac0:EvseS  :: Session (run_id=2BECC771454FC714, ev_mac=00:01:87:05:A0:84): received CM_MNBC_SOUND_IND
````