EVerest/EVerest #1024: Auth module crashes when validate_token timesout

wku12, 2025-01-24:
### Describe the bug

It seems that EverestTimeoutError is not handled in Auth module:

```
Jan 24 10:51:23 everest-manager[23420]: {"SEVERITY": "ERRO", "SYSLOG_IDENTIFIER": "auth:Auth      ", "CODE_FILE": "", "CODE_LINE": "", "MESSAGE": "Timeout while waiting for result of xxx->token_validator:auth_token_validator->validate_token()"}
Jan 24 10:51:23 everest-manager[23420]: terminate called after throwing an instance of 'boost::wrapexceptEverest::EverestTimeoutError'
Jan 24 10:51:23 everest-manager[23420]:  what():  Timeout while waiting for result of xxx->token_validator:auth_token_validator->validate_token()
Jan 24 10:51:26 systemd-coredump[24275]: [🡕] Process 23420 (auth:Auth) of user 0 dumped core.
Jan 24 10:51:26 everest-manager[23390]: {"SEVERITY": "CRIT", "SYSLOG_IDENTIFIER": "manager        ", "CODE_FILE": "", "CODE_LINE": "", "MESSAGE": "Module auth (pid: 23420) exited with status: 134. Terminating all modules."}
```

### EVerest Domain

Authorization

### Affected EVerest Module

Auth

### To Reproduce

Steps to reproduce:

1. Call validate_token and do not reply within timeout

hikinggrass, 2025-01-24:
Thanks for raising this issue, that's indeed something that needs addressing. There's [a slightly related PR in everest-framework](https://github.com/EVerest/everest-framework/pull/210) that makes handling the exceptions that can occur during command calls a bit more explicit but even then the handling ultimately needs to be done in the caller. This would mean having try-catch blocks around command calls and the appropriate error handling/recovering logic.

mlitre, 2025-04-10:
Closed by #1032

closed 2025-04-10 by mlitre