EVerest/EVerest #1044: How to add third external energy limit source in config file

dev-enphase, 2025-02-05:
[config_file.txt](https://github.com/user-attachments/files/18672803/config_file.txt)

Hi Team,

I am facing issue in adding third external energy source to set external limit using energy manager module. I am able to add two external energy sources and able to start/stop charge based on the available power but not able add another external energy source. We have one energy node for OCPP and Two for external source.

Can you please help me in create a right energy tree for third external energy source? I have attached the config file which I edited and not working for third energy source. Below is the module which require evse_manager_2_external_sink to set external limit. Out of three sources lowest limit should get applied.

 local_green_charging:
    module: LocalGreenChargingManager
    connections:
      store:
        - module_id: persistent_store
          implementation_id: main
      power_interface:
        - module_id: bsp
          implementation_id: powermeter
      evse_energy_sink:
        - module_id: evse_manager_2_external_sink
          implementation_id: external_limits
      evse_manager:
        - module_id: connector_1
          implementation_id: evse

dev-enphase, 2025-02-05:
@corneliusclaussen  Can you please check the config file and suggest what is missing and how to add  third external energy source?

corneliusclaussen, 2025-02-05:
They are not configured in a chain, both external energy nodes connect to the ocpp node

dev-enphase, 2025-02-05:
@corneliusclaussen  I didn't get you.

  I have connected both external_1 and external_2 to OCPP node

evse_manager_1_ocpp_sink:
    module: EnergyNode
    mapping:
      module:
        evse: 1
    config_module:
      fuse_limit_A: 32.0
      phase_count: 3
    connections:
      energy_consumer:
        - module_id: connector_1
          implementation_id: energy_grid

  evse_manager_1_external_sink:
    module: EnergyNode
    mapping:
      module:
        evse: 1
    config_module:
      fuse_limit_A: 32.0
      phase_count: 3
    connections:
      energy_consumer:
        - module_id: evse_manager_1_ocpp_sink
          implementation_id: energy_grid

  evse_manager_2_external_sink:
    module: EnergyNode
    mapping:
      module:
        evse: 1
    config_module:
      fuse_limit_A: 32.0
      phase_count: 3
    connections:
      energy_consumer:
        - module_id: evse_manager_1_ocpp_sink
          implementation_id: energy_grid

Can you please correct me what is wrong here?