EVerest/EVerest #1071: Dockerized charging station issue - systemd PID 1

AviranAbady, 2025-03-03:
Hey team,

I'm trying to dockerize a EVerest charging station, I'm able to complete the build process as described in https://github.com/EVerest/everest-core/blob/main/README.md

You can see the Dockerfile [here](https://github.com/AviranAbady/tzi-charging-station/tree/main).

When running the `run-sil.sh` script, the system seems to load up to the point of ` 🌀🌀🌀 Ready to start charging 🌀🌀🌀`

And from there im getting `System has not been booted with systemd as init system (PID 1). Can't operate.` errors

You can view the [full log here](https://github.com/AviranAbady/tzi-charging-station/blob/main/error.txt)

`systemd` is not part of the docker image (ubuntu 22.04), so the error make sense.

Im trying understand the source of the error, and hopefully find a way around it, which system service is actually needed here? trying to be accessed by finding `systemd` at PID 1?

Any other ideas of what goes wrong in the execution?
Suggestions on how to debug?

hikinggrass, 2025-03-03:
That's likely from the _Setup_ module which uses systemctl to query some information (eg you can see that in the code here: https://github.com/EVerest/everest-core/blob/b6c188f86f8c6100828d949e0717a357f184d8eb/modules/Setup/Setup.cpp#L263 ).
That module isn't really necessary and it's functionality can certainly be implemented without relying on systemd (patches/PRs are very welcome here 😃 ), so you can just remove it from your config and these errors should disappear

AviranAbady, 2025-03-04:
Thank you for the quick response @hikinggrass , `publish_ap_state()` is indeed causing the issue.
I can PR and replace the use of `systemctl` with `pgrep` which is more likely to be found in linux distros but the same issue will occur on minimal linux distributions which may omit pgrep as well. 

The real issue comes from `run_application()` which returns 1 when the executable is not found which in some cases should not be considered as a failed execution, I assume modifying run_application() should be discussed a bit more extensively.

closed 2025-10-21 by Pietfried