EVerest/EVerest #1072: Does not compile with boost library 1.87.0

s7rul, 2025-03-04:
### Describe the bug

This is because the `expires_from_now` method was removed from the `basic_waitable_timer` template class (marked deprecated in version 1.66.0). This can be fixed by using the `expire_after` method instead for all boost versions from 1.66.0.

### EVerest Domain

Compilation

hikinggrass, 2025-03-06:
Thanks for reporting this, should be a straightforward fix in [libtimer](https://github.com/EVerest/libtimer/blob/173e96e8ea78ef3918371db02474b64494329e6e/include/everest/timer.hpp#L129)

andistorm, 2025-03-06:
Will be fixed with https://github.com/EVerest/libtimer/pull/12

closed 2025-03-12 by andistorm