EVerest/EVerest #1292: ocpp201-pnc config crashes after upgrade from 2025.3.0 to 2025.6.0

catarial, 2025-08-05:
### Describe the bug

Using the default `config-sil-ocpp201-pnc.yaml` file and Maeve cams results in multiple errors in the EvseV2G and PyEvJosev modules related to certificates. The default AC and DC configs work fine and have no errors.

<details>

<summary>
everest logs
</summary>

```
```
2025-08-05 20:16:32.086108 [INFO] manager          ::   ________      __                _
2025-08-05 20:16:32.086323 [INFO] manager          ::  |  ____\ \    / /               | |
2025-08-05 20:16:32.086332 [INFO] manager          ::  | |__   \ \  / /__ _ __ ___  ___| |_
2025-08-05 20:16:32.086335 [INFO] manager          ::  |  __|   \ \/ / _ \ '__/ _ \/ __| __|
2025-08-05 20:16:32.086338 [INFO] manager          ::  | |____   \  /  __/ | |  __/\__ \ |_
2025-08-05 20:16:32.086342 [INFO] manager          ::  |______|   \/ \___|_|  \___||___/\__|
2025-08-05 20:16:32.086345 [INFO] manager          ::
2025-08-05 20:16:32.086348 [INFO] manager          :: everest-framework 0.22.1 @v0.22.1
2025-08-05 20:16:32.086353 [INFO] manager          :: everest-core_tests  @2025.6.0
2025-08-05 20:16:32.086360 [INFO] manager          ::
2025-08-05 20:16:32.086363 [INFO] manager          :: Using MQTT broker mqtt-server:1883
2025-08-05 20:16:32.090486 [INFO] everest_ctrl     :: Launching controller service on port 8849
2025-08-05 20:16:32.109930 [INFO] manager          :: Boot mode is set to YamlFile, loading module configs from YAML file
2025-08-05 20:16:32.110010 [INFO] manager          :: Loading config file at: /ext/source/config/config-sil-ocpp201-pnc.yaml
2025-08-05 20:16:32.271436 [INFO] manager          :: Config loading completed in 184ms
2025-08-05 20:16:32.291746 [INFO] manager          :: Starting 23 modules
2025-08-05 20:16:34.648882 [INFO] api:API          :: Module api initialized [2320ms]
2025-08-05 20:16:34.651368 [INFO] ev_api:EvAPI     :: Module ev_api initialized [2319ms]
2025-08-05 20:16:34.749130 [INFO] grid_connection  :: Module grid_connection_point initialized [2303ms]
2025-08-05 20:16:34.753012 [INFO] error_history:E  :: Resetting database
2025-08-05 20:16:34.754786 [INFO] evse_manager_2_  :: Module evse_manager_2_api_sink initialized [2307ms]
2025-08-05 20:16:34.773166 [INFO] error_history:E  :: Module error_history initialized [2449ms]
2025-08-05 20:16:34.798195 [INFO] ev_manager_1:Ev  :: Module ev_manager_1 initialized [2462ms]
2025-08-05 20:16:34.827463 [INFO] system:System    :: Module system initialized [2198ms]
2025-08-05 20:16:34.844971 [INFO] energy_manager:  :: Module energy_manager initialized [2520ms]
2025-08-05 20:16:34.867223 [INFO] slac:SlacSimula  :: Module slac initialized [2243ms]
2025-08-05 20:16:34.868690 [INFO] ev_manager_2:Ev  :: Module ev_manager_2 initialized [2527ms]
2025-08-05 20:16:34.873536 [INFO] evse_security:E  :: Module evse_security initialized [2346ms]
2025-08-05 20:16:34.924346 [INFO] evse_manager_1:  :: Module evse_manager_1 initialized [2388ms]
2025-08-05 20:16:34.928837 [INFO] yeti_driver_2:Y  :: Module yeti_driver_2 initialized [2206ms]
2025-08-05 20:16:34.929018 [INFO] evse_manager_1_  :: Module evse_manager_1_api_sink initialized [2586ms]
2025-08-05 20:16:34.935907 [INFO] evse_manager_2_  :: Module evse_manager_2_ocpp_sink initialized [2492ms]2025-08-05 20:16:34.935918 [INFO] evse_manager_1_  :: Module evse_manager_1_ocpp_sink initialized [2502ms]

2025-08-05 20:16:34.937108 [INFO] auth:Auth        :: Module auth initialized [2599ms]
2025-08-05 20:16:34.952966 [INFO] yeti_driver_1:Y  :: Module yeti_driver_1 initialized [2230ms]
2025-08-05 20:16:35.036185 [INFO] token_provider_  :: Module token_provider_1 initialized [2405ms]
2025-08-05 20:16:35.040208 [INFO] iso15118_charge  :: Module iso15118_charger initialized [2315ms]
2025-08-05 20:16:35.042657 [INFO] evse_manager_2:  :: Module evse_manager_2 initialized [2417ms]
2025-08-05 20:16:35.043565 [INFO] ocpp:OCPP201     :: Module ocpp initialized [2209ms]
2025-08-05 20:16:35.816015 [INFO] iso15118_car     :: Module iso15118_car initialized [2033ms]
2025-08-05 20:16:35.856695 [INFO] manager          :: Clearing retained topics published by manager during startup
2025-08-05 20:16:35.857341 [INFO] manager          :: 🚙🚙🚙 All modules are initialized. EVerest up and running [3776ms] 🚙🚙🚙

2025-08-05 20:16:35.862796 [INFO] iso15118_charge  :: TCP server on eth0 is listening on port [fe80::949a:ff:fe88:3477%11]:61341

2025-08-05 20:16:35.862983 [INFO] iso15118_charge  :: TLS server on eth0 is listening on port [fe80::949a:ff:fe88:3477%11]:64109
2025-08-05 20:16:35.906672 [INFO] evse_security:E  :: Requesting leaf certificate info: V2G
2025-08-05 20:16:35.913594 [INFO] evse_security:E  :: Found valid leaf: ["/ext/dist/etc/everest/certs/client/cso/CPO_CERT_CHAIN.pem"]
2025-08-05 20:16:35.914275 [INFO] evse_security:E  :: Found valid leaf: ["/ext/dist/etc/everest/certs/client/cso/SECC_LEAF.pem"]
2025-08-05 20:16:35.961677 [WARN] iso15118_charge void {anonymous}::log_handler(openssl::log_level_t, const std::string&) :: <n> certificates != <n> OCSP responses

2025-08-05 20:16:35.990200 [INFO] iso15118_charge  :: SDP socket setup succeeded
2025-08-05 20:16:36.032378 [INFO] ocpp:OCPP201     :: Target version: 3, current version: 0
2025-08-05 20:16:36.033592 [INFO] ocpp:OCPP201     :: Migration files to apply:
2025-08-05 20:16:36.033624 [INFO] ocpp:OCPP201     :: Migration file [up] version 1, path: /ext/dist/share/everest/modules/OCPP201/device_model_migrations/1_up-initial.sql
2025-08-05 20:16:36.033639 [INFO] ocpp:OCPP201     :: Migration file [up] version 2, path: /ext/dist/share/everest/modules/OCPP201/device_model_migrations/2_up-variable_source.sql
2025-08-05 20:16:36.033643 [INFO] ocpp:OCPP201     :: Migration file [up] version 3, path: /ext/dist/share/everest/modules/OCPP201/device_model_migrations/3_up-variable_required.sql
2025-08-05 20:16:36.054667 [INFO] ocpp:OCPP201     :: Established connection to device model database: "/ext/dist/share/everest/modules/OCPP201/device_model_storage.db"
[... 119 lines cut ...]
2025-08-05 20:17:45.273252 [ERRO] iso15118_car    pybind11_init_everestpy(pybind11::module_&)::<lambda(const std::string&)> :: EXI message (ns=Namespace.ISO_V2_MSG_DEF) where validation failed: 80980239d117d97fdffbd9906082002400402000020000201000

2025-08-05 20:17:50.279414 [ERRO] iso15118_charge void dlog_func(dloglevel_t, const char*, int, const char*, const char*, ...) :: connection_read(header) failed: Success

2025-08-05 20:17:50.279556 [ERRO] iso15118_charge void dlog_func(dloglevel_t, const char*, int, const char*, const char*, ...) :: v2g_incoming_v2gtp() (previous message "Certificate Installation") failed

2025-08-05 20:17:50.279626 [INFO] iso15118_charge  :: v2g_dispatch_connection exited with -1
2025-08-05 20:17:50.280523 [ERRO] iso15118_car    pybind11_init_everestpy(pybind11::module_&)::<lambda(const std::string&)> :: EVCC tried to initiate a V2GCommunicationSession, but maximum number of SDP retry cycles (1) is now reached. Shutting down high-level communication. Unplug and plug in the cable again if you want to start anew.
Traceback (most recent call last):
  File "/ext/dist/libexec/everest/modules/PyEvJosev/../../3rd_party/josev/iso15118/evcc/comm_session_handler.py", line 572, in get_from_rcv_queue
    await self.restart_sdp(True)
  File "/ext/dist/libexec/everest/modules/PyEvJosev/../../3rd_party/josev/iso15118/evcc/comm_session_handler.py", line 389, in restart_sdp
    raise SDPFailedError(
iso15118.shared.exceptions.SDPFailedError: EVCC tried to initiate a V2GCommunicationSession, but maximum number of SDP retry cycles (1) is now reached. Shutting down high-level communication. Unplug and plug in the cable again if you want to start anew.
```

</details>

### EVerest Domain

OCPP2.0.1

### To Reproduce

Run Everest using the default `config-sil-ocpp201-pnc.yaml` config connected to the Maeve csms. Issue the `iso_start_v2g_session AC` command to Everest.

Pietfried, 2025-08-06:
@shankari The EV is requesting `CertificateInstall` to install a contract certificate in the EV, so this request is forwarded to the CSMS to retrieve the contract certificate. If you want to disable that the EV requests a contract certificate installation, you can do this using the [is_cert_install_needed](https://github.com/EVerest/everest-core/blob/main/modules/PyEvJosev/manifest.yaml#L33) configuration parameter of [PyEvJosev](https://github.com/EVerest/everest-core/tree/main/modules/PyEvJosev).

`ContractCertificateInstallationEnabled` is an optional property of the `ISO15118Ctrlr`. If it is not present or doesn't have a value, it defaults to false like all other optional configuration, that is not set or does not have a value. The standardized components provided in libocpp only serve as an example and should be configured according to the requirements and composition of the charging station.

catarial, 2025-08-06:
I set the value to true and that warning went away but I'm still getting the other errors

<details>

<summary>
everest logs
</summary>

```
2025-08-06 16:07:35.080894 [INFO] manager          ::   ________      __                _
2025-08-06 16:07:35.081215 [INFO] manager          ::  |  ____\ \    / /               | |
2025-08-06 16:07:35.081257 [INFO] manager          ::  | |__   \ \  / /__ _ __ ___  ___| |_
2025-08-06 16:07:35.081288 [INFO] manager          ::  |  __|   \ \/ / _ \ '__/ _ \/ __| __|
2025-08-06 16:07:35.081314 [INFO] manager          ::  | |____   \  /  __/ | |  __/\__ \ |_
2025-08-06 16:07:35.081341 [INFO] manager          ::  |______|   \/ \___|_|  \___||___/\__|
2025-08-06 16:07:35.081360 [INFO] manager          ::
2025-08-06 16:07:35.081382 [INFO] manager          :: everest-framework 0.22.1 @v0.22.1
2025-08-06 16:07:35.081410 [INFO] manager          :: everest-core_tests  @2025.6.0
2025-08-06 16:07:35.081435 [INFO] manager          ::
2025-08-06 16:07:35.081463 [INFO] manager          :: Using MQTT broker mqtt-server:1883
2025-08-06 16:07:35.104580 [INFO] everest_ctrl     :: Launching controller service on port 8849
2025-08-06 16:07:35.129150 [INFO] manager          :: Boot mode is set to YamlFile, loading module configs from YAML file
2025-08-06 16:07:35.129206 [INFO] manager          :: Loading config file at: /ext/source/config/config-sil-ocpp201-pnc.yaml
2025-08-06 16:07:35.312164 [INFO] manager          :: Config loading completed in 228ms
2025-08-06 16:07:35.316353 [INFO] manager          :: Starting 23 modules
2025-08-06 16:07:37.915205 [INFO] ev_manager_2:Ev  :: Module ev_manager_2 initialized [2503ms]
2025-08-06 16:07:37.945876 [INFO] ev_manager_1:Ev  :: Module ev_manager_1 initialized [2579ms]
2025-08-06 16:07:38.030934 [INFO] ocpp:OCPP201     :: Module ocpp initialized [2370ms]
2025-08-06 16:07:38.041209 [INFO] auth:Auth        :: Module auth initialized [2686ms]
2025-08-06 16:07:38.047960 [INFO] ev_api:EvAPI     :: Module ev_api initialized [2686ms]
2025-08-06 16:07:38.048580 [INFO] energy_manager:  :: Module energy_manager initialized [2687ms]
2025-08-06 16:07:38.078876 [INFO] yeti_driver_1:Y  :: Module yeti_driver_1 initialized [2326ms]2025-08-06 16:07:38.079092 [INFO] evse_manager_1_  :: Module evse_manager_1_api_sink initialized [2705ms]
2025-08-06 16:07:38.081267 [INFO] iso15118_charge  :: Module iso15118_charger initialized [2419ms]

2025-08-06 16:07:38.082899 [INFO] grid_connection  :: Module grid_connection_point initialized [2615ms]
2025-08-06 16:07:38.088918 [INFO] api:API          :: Module api initialized [2736ms]
2025-08-06 16:07:38.105549 [INFO] error_history:E  :: Resetting database
2025-08-06 16:07:38.151388 [INFO] evse_manager_2_  :: Module evse_manager_2_api_sink initialized [2690ms]
2025-08-06 16:07:38.160892 [INFO] system:System    :: Module system initialized [2509ms]
2025-08-06 16:07:38.161241 [INFO] evse_manager_1_  :: Module evse_manager_1_ocpp_sink initialized [2712ms]
2025-08-06 16:07:38.161730 [INFO] error_history:E  :: Module error_history initialized [2806ms]
2025-08-06 16:07:38.162209 [INFO] evse_manager_2_  :: Module evse_manager_2_ocpp_sink initialized [2616ms]
2025-08-06 16:07:38.165496 [INFO] yeti_driver_2:Y  :: Module yeti_driver_2 initialized [2417ms]
2025-08-06 16:07:38.201690 [INFO] evse_manager_1:  :: Module evse_manager_1 initialized [2541ms]
2025-08-06 16:07:38.229984 [INFO] token_provider_  :: Module token_provider_1 initialized [2574ms]
2025-08-06 16:07:38.245679 [INFO] evse_manager_2:  :: Module evse_manager_2 initialized [2584ms]
2025-08-06 16:07:38.252199 [INFO] evse_security:E  :: Module evse_security initialized [2577ms]
2025-08-06 16:07:38.257321 [INFO] slac:SlacSimula  :: Module slac initialized [2597ms]
2025-08-06 16:07:39.264318 [INFO] iso15118_car     :: Module iso15118_car initialized [2313ms]
2025-08-06 16:07:39.266568 [INFO] manager          :: Clearing retained topics published by manager during startup
2025-08-06 16:07:39.268194 [INFO] manager          :: 🚙🚙🚙 All modules are initialized. EVerest up and running [4191ms] 🚙🚙🚙

2025-08-06 16:07:39.271448 [INFO] iso15118_charge  :: TCP server on eth0 is listening on port [fe80::d83d:f1ff:fe86:5ce3%11]:61341

2025-08-06 16:07:39.271559 [INFO] iso15118_charge  :: TLS server on eth0 is listening on port [fe80::d83d:f1ff:fe86:5ce3%11]:64109
2025-08-06 16:07:39.280542 [INFO] evse_security:E  :: Requesting leaf certificate info: V2G
2025-08-06 16:07:39.335619 [INFO] evse_security:E  :: Found valid leaf: ["/ext/dist/etc/everest/certs/client/cso/CPO_CERT_CHAIN.pem"]
2025-08-06 16:07:39.336628 [INFO] evse_security:E  :: Found valid leaf: ["/ext/dist/etc/everest/certs/client/cso/SECC_LEAF.pem"]
2025-08-06 16:07:39.429924 [WARN] iso15118_charge void {anonymous}::log_handler(openssl::log_level_t, const std::string&) :: <n> certificates != <n> OCSP responses
2025-08-06 16:07:39.448225 [INFO] ocpp:OCPP201     :: Target version: 3, current version: 0
2025-08-06 16:07:39.462367 [INFO] ocpp:OCPP201     :: Migration files to apply:
2025-08-06 16:07:39.462409 [INFO] ocpp:OCPP201     :: Migration file [up] version 1, path: /ext/dist/share/everest/modules/OCPP201/device_model_migrations/1_up-initial.sql
2025-08-06 16:07:39.462419 [INFO] ocpp:OCPP201     :: Migration file [up] version 2, path: /ext/dist/share/everest/modules/OCPP201/device_model_migrations/2_up-variable_source.sql
[... cut ...]
  File "/ext/dist/libexec/everest/modules/PyEvJosev/../../3rd_party/josev/iso15118/evcc/comm_session_handler.py", line 572, in get_from_rcv_queue
    await self.restart_sdp(True)
  File "/ext/dist/libexec/everest/modules/PyEvJosev/../../3rd_party/josev/iso15118/evcc/comm_session_handler.py", line 389, in restart_sdp
    raise SDPFailedError(
iso15118.shared.exceptions.SDPFailedError: EVCC tried to initiate a V2GCommunicationSession, but maximum number of SDP retry cycles (1) is now reached. Shutting down high-level communication. Unplug and plug in the cable again if you want to start anew.
```

</details>

Pietfried, 2025-08-11:
Can you see if the CertificateInstallation request is sent via OCPP2.x in the `Get15118EVCertificateRequest` message?

Pietfried, 2025-08-13:
> I don't know how to do that

You should be able to see this in the OCPP logs. By default these are located in `/tmp/everest_ocpp_logs/`

catarial, 2025-10-16:
No, I'm not able to reproduce the issue anymore